ian-ross/kicad-plugins #1: installing problems

insidecircuits, 2020-09-08:
Hi,
I am not able to install this on either windows or linux, I copied it in `kicad/scripting/plugins` alongside other plugins but it is not recognized by KiCad, I am using version 5.1.6 and also tried with the current nightly build.

I see that other plugins have a different file structure, not sure if this is the problem.

ian-ross, 2020-09-08:
That's weird. You should be able just to put the single `SchematicPositionsToLayout.py` file in your KiCad plugins directory. It's a simple single-file plugin: you've probably seen the more complex setup for multi-file plugins. I guess it won't work with the nightlies because the schematic file format has changed, but it definitely works with 5.1.6.

Are you putting the plugin script into the per-user plugin directory? I can't tell from your report exactly where you're putting it. (That would be `~/.kicad_plugins` on Linux, and somewhere like `C:\Users\username\AppData\Roaming\kicad\scripting\plugins` on Windows.)

insidecircuits, 2020-09-09:
Well I copied it in `~/.kicad_plugins` and also `/usr/share/kicad/scripting/plugins` and I still can't see it in KiCad 5.1.6
It's strange.

ian-ross, 2020-09-09:
It is strange. Did you restart KiCad or do "Tools -> External Plugins... -> Refresh Plugins" after putting the Python file into the plugins directory? You need to do one of those two things for the plugin to appear in the menu.

Do you have any other plugins installed? If not, you could try installing something else to see if the problem is with your KiCad setup or something weird about my plugin (the interactive HTML BOM plugin is a good one to test with: https://github.com/openscopeproject/InteractiveHtmlBom).

insidecircuits, 2020-09-09:
I have copied the InteractiveHtmlBom folder to `~/.kicad_plugins` and hit Refresh Plugins without restarting and it appeared in the menu and toolbar.

ian-ross, 2020-09-09:
OK, that's really strange then, because if I copy the `SchematicPositionsToLayout.py` file to `~/.kicad_plugins`, it works the same as the interactive HTML BOM plugin, i.e. it shows up in the plugins menu after I do "Refresh Plugins".

Just to check one last thing: are you copying *just* the `SchematicPositionsToLayout.py` file into your `~/.kicad_plugins`, and not the whole of my `kicad-plugins` repository directory? You need to have the plugin file as `~/.kicad_plugins/SchematicPositionsToLayout.py`, *not* as `~/.kicad_plugins/kicad-plugins/SchematicPositionsToLayout.py`, which is what you'd get it you just cloned by plugins repository into your `~/.kicad_plugins`. (Sorry if this is telling you something you already know, but I can't work out why it's not working for you otherwise!)

insidecircuits, 2020-09-09:
Yes I have it like that ~/.kicad_plugins/SchematicPositionsToLayout.py
I am wondering if it's not a issue of Python 2 / 3 because I think I have both installed, not sure if ti can be an issue.

ian-ross, 2020-09-09:
I *believe* that the plugin code should work with both Python 3 and Python 2, but Python version problems are definitely a possibility.

What does your version information in Pcbnew look like? (Go to "Help -> About KiCad" and click "Show Version Info" in the dialog.) Mine looks like this:

```
Application: Pcbnew
Version: 5.1.6, release build
Libraries:
    wxWidgets 3.0.5
    libcurl/7.72.0 OpenSSL/1.1.1g zlib/1.2.11 zstd/1.4.5 libidn2/2.3.0 libpsl/0.21.1 (+libidn2/2.3.0) libssh2/1.9.0 nghttp2/1.41.0
Platform: Linux 5.4.61-1-lts x86_64, 64 bit, Little endian, wxGTK
Build Info:
    wxWidgets: 3.0.5 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
    Boost: 1.72.0
    OpenCASCADE Technology: 7.4.0
    Curl: 7.70.0
    Compiler: GCC 9.3.0 with C++ ABI 1013

Build settings:
    USE_WX_GRAPHICS_CONTEXT=OFF
    USE_WX_OVERLAY=ON
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=ON
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=ON
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=OFF
    KICAD_USE_OCC=ON
    KICAD_SPICE=ON
```

The most relevant thing is that `KICAD_SCRIPTING_PYTHON3` flag.

DanielMartensson, 2020-10-14:
I have the same issue.

Placing here `/home/myUser/.kicad_plugins/` works. Not `/usr/share/kicad/plugins/`

gabrieled07, 2021-03-19:
Hi, I've had the same issue
I've downloaded the .py file directly from github (lazy me) and placed into /usr/share/kicad/plugins on kicad 5.1.9 running on Ubuntu 20.09: it was not detected.
A little research led me here.

It turns out, downloading the file directly from the site include line numbers and other things, I didn't knew this, and didn't check it after download.
Properly downloading the file, by git clone or zip method and placing at ~./kicad_plugins/ fix the -non existant- issue.
By the way, ~/kicad_plugins is not created when installed kicad with #apt install kicad on said ubuntu 20.04 linux.

Sorry if this is not related, it might not be insidecircuits issue, and although is too obvious it was what happened to me.

Your script is awesome, huge huge time saver.

best regards,
Gabriel

ian-ross, 2021-03-21:
@gabrieled07 Thanks for the report, and I'm glad you find the plugin useful!

bombela, 2021-09-09:
This path is what worked for me: `~/.config/kicad/scripting/plugins/`

alchemyyy, 2021-11-01:
I had this issue and figured out that it was something in the tuple map version parser on line 11 causing it. I deleted that entire version check brick and simply stated ENABLE_KICAD_V6_API=True as I am on 5.99 right now.

closed 2023-03-03 by ian-ross