ian-ross/kicad-plugins #2: Mainline Support

nitz, 2020-12-30:
Hello Ian!

Fantastic plugin you've got here. I actually stumbled across it while searching for a different plugin, but am quite glad I did. I felt like what your plugin does here should honestly be (even if just optionally,) the default behavior when updating the PCB from the schematic.

As such, I thought to go add a feature request on the KiCAD issue tracker, only to first find one that mentions your plugin: https://gitlab.com/kicad/code/kicad/-/issues/2405

The interesting bit, is the most recent comment from [Seth](https://gitlab.com/kicad/code/kicad/-/issues/2405#note_402147492):

> It's a neat plugin.  If they wanted to submit a patch in C++, we could work with them to get it incorporated.

I don't know how much C++ you do, or if you're up to such a thing, but it sounds like there'd be welcome support for your feature in the mainline!

ian-ross, 2020-12-30:
Thanks for the heads-up! I hadn't seen that KiCad issue at all. I'll have a think about whether I want to get involved in writing a patch. I'm OK with C++, but I've not done any development on the core of KiCad before, so there would be a learning curve there.

nitz, 2020-12-31:
Awesome! I haven't either (in fact, cloned the source for the first time today) and am just working on getting set up to build it myself. I very much know that feeling of whether or not it's worth taking on a new project like that!

ian-ross, 2023-03-03:
I'm going to close this one, because I am definitely not going to have time to do it, and in addition, KiCad 7 has a new "Pack and Move Footprints" feature that makes the initial schematic ⇒ board transition much easier. See here for a video demo: https://gitlab.com/kicad/code/kicad/-/merge_requests/1325

closed 2023-03-03 by ian-ross