ian-ross/kicad-plugins #3: The problem when starting the plugin

ifadiga, 2021-02-27:
I have multiple errors when starting the plugin in KiCad.
My version is r20816.5079881f3f which is a nightly KiCad.
From the previous issue, I read that plugin not working in nightly KiCad but did you planning to add support for it?
I can't use stable release because using other plugins which not working on that release.

ian-ross, 2021-02-27:
I'll update it when v6 is released. There's still work going on on the Python API, so I don't want to update until that's stabilised. Because v5 doesn't have a schematic API, this plugin works by parsing the schematic file. I understand that v6 is going to have a schematic API, which will make doing this sort of thing much more reliable.

ifadiga, 2021-02-28:
Thanks for the explanation,
I will wait for v6 and your update for the plugin.

wm93, 2021-07-04:
Hey Ian,

the schematic API seems to be an open point for v6 until today. I tried to modify your plugin for v6 to get similar results as your original v5 code. There're probably some bugs and the code can be improved a lot but at least it seems to do something. Would you consider checking out my code and maybe post a version for the nightly builds v599?

ian-ross, 2021-07-06:
@wm93 Sure, I'll take a look -- do you have the code on GH somewhere? I wasn't planning to do anything with this until there's an official v6 release candidate, but maybe we can get it working before then. I've not been following the schematic API development, but if it's ready, I'm happy to see what's up. (Is the schematic API documented anywhere yet?)

wm93, 2021-07-10:
> Sure, I'll take a look -- do you have the code on GH somewhere?

Invited you to the repository. Made it private for now, since this is just an intermediate workaround until the eeschema API will be released (and maybe because I don't want others to view my bad code ;) )

ian-ross, 2021-07-11:
@wm93 I misunderstood what you said above. I thought the v6 schematic API had been finalised and you'd written code to use it. (My actual thinking: "Oh wow, that's great! Someone's already done it so I won't have to!") I see that that's not the case and, as you said, your solution is an intermediate one. It's pretty encouraging that it's still quite short, even though you need to parse the S-expression schematic file format yourself!

I'm very happy to include your code in this repository as an "at your own risk" option for v599 users until v6 is out and we can update to the new schematic API. I'm not able to test it myself, as I don't have v599 installed, but if you say that it works, I'm perfectly happy to accept a pull request to include it here!

HaydenHu, 2021-08-13:
> > 当然，我去看看——你有 GH 的代码吗？
> 
> 邀请您访问存储库。暂时将其设为私有，因为在 eeschema API 发布之前这只是一个中间解决方法（也许是因为我不希望其他人查看我的错误代码；））

Can you share your code?

ian-ross, 2021-09-27:
@lvsoft has very kindly added support for the new schematic file format.

closed 2021-09-27 by ian-ross