ian-ross/kicad-plugins #4: Can you add support for 5.99?

HaydenHu, 2021-08-07:
I like this plug-in very much. Thank you very much for your work! However, it does not support 5.99 at present, and the new schematic API is likely not to be added. However, V6 cannot be released for a long time, and the release date is still unpredictable. I hope you can increase your support for 5.99

ian-ross, 2021-08-08:
This is a duplicate of #3. I really don't want to do this, as I'm not using 5.99 myself. I've not been keeping up with what's happening with the current development of v6, and was definitely planning to wait for the schematic API to be released before updating this. As I said on the other issue, I'm happy to accept PRs to handle this. If you're really impatient, @wm93 has some changes in a private repository that might do the job for you — you could ask them if you can use their code.

But I **will** update the plugin when v6 is released with the new schematic API!

ian-ross, 2021-09-27:
@lvsoft has very kindly added support for the new schematic file format.

closed 2021-09-27 by ian-ross