ian-ross/kicad-plugins #12: Update for KiCad 7

m3rkator, 2023-03-01:
Hello Ian

Thanks for your wonderful plugin, which made my work so much easier in the past.

Unfortunately, it stopped working with KiCad 7

[image]

Screenshot inline.

Can you do something about it?

Bye
Holger

ian-ross, 2023-03-01:
There's a KiCad 7?! That shows how up to date I am with KiCad... I had no idea there was even going to be a new version.

Let me see what I can do.

ian-ross, 2023-03-03:
@m3rkator Do you think you could try the latest `SchematicPositionsToLayout.py`? I've made some updates for KiCad 7 that I think should do the trick. I've tested with a couple of my own projects, and it seems OK (including hierarchical sheets), but it's hard to be 100% confident. Thanks!

m3rkator, 2023-03-03:
Hello  @ian-ross 

Same here. Update works with 100% of all my projects. But i can't speak for the rest of the globe.

For myself i can say: Awesome and many thanks.

Bye
Holger

ian-ross, 2023-03-04:
Thanks for checking that out, @m3rkator !

closed 2023-03-04 by ian-ross