ian-ross/kicad-plugins #13: Doesn't work with Kicad 7

Pkemarco, 2023-03-16:
Hi guys and thank you for your hard work.
I tried recently to use this tool with Kicad 7 but I am getting this error :

[image]

Can you help me to fix it please ?
Thanks !

ian-ross, 2023-03-16:
Interesting. I thought I'd got this working for KiCad 7 a couple of weeks ago, but it looks like I didn't deal with hierarchical sheets properly. (I thought I had...) I'll take a look later today. It should be a simple fix, I think.

Pkemarco, 2023-03-16:
Ok, thanks !!

Pkemarco, 2023-03-16:
It works now ! Thanks you !!!

Pkemarco, 2023-03-16:
> Hi guys and thank you for your hard work. I tried recently to use this tool with Kicad 7 but I am getting this error :
> 
> [image]
> 
> Can you help me to fix it please ? Thanks !

closed 2023-03-16 by Pkemarco